KiCad/kicad-library #1674: NXP mcu libraries

t3hcatpaw, 2017-09-30:
Hi all, hi @SchrodingersGat, hi @myfreescalewebpage,

recently, @SchrodingersGat has merged PRs #359 and #1540 independently. As a result, Kinetis devices are now scattered over the two libs `nxp_armmcu` (together with LPC devices) and `nxp_kinetis` with several KL0x devices being part of both libs at the same time (duplicates).

I therefore suggest
1. we - @myfreescalewebpage and I - team up to merge all Kinetis parts into the newly created dedicated `nxp_kinetis` lib. I would be happy to issue a PR for that.
2. `nxp_armmcu`, which afterwards should only contain LPC devices, be consequently renamed to `nxp_lpc`.

I remember a discussion (issue?) with a list of how symbol libraries should be renamed, but couldn't find it anymore. Maybe a part of this issue has already been solved there.

Best regards,
Dennis

SchrodingersGat, 2017-09-30:
You will find that discussion here: https://github.com/KiCad/kicad-library/issues/1402

There are also new capitalisation conventions which will come in to play for KLC V3. 

So, I would suggest that your changes are forward compatible with that. 

New library naming scheme is to put functionality first. e.g.

* MCU_NXP_Kinetis
* MCU_Microchip_PIC18

If you want to submit a PR arranging the NXP micros as above, that'd be great.

t3hcatpaw, 2017-10-02:
Thank you! Libs are renamed accordingly, symbols moved.

@myfreescalewebpage Would you mind if I 'overwrite' your KL0x symbols?

myfreescalewebpage, 2017-10-03:
Well be sure I will be a little bit disappointed because it took about 1 year to get my PR accepted with all moves in the KLC !!
However, it's open, but we should keep the style of all the NXP MCU symbols consistent, for example with the S08 family.
How many devices to you have to overwrite and how many do are you adding I have not provided ?

t3hcatpaw, 2017-10-03:
> Well be sure I will be a little bit disappointed because it took about 1 year to get my PR accepted with all moves in the KLC !!

#359 was great work - I'm sorry if I upset you and please be assured I'm trying to be as careful as possible to not undo your effort, just enhance it. :grin: 

First of all, I won't touch your massive S08 family addition, just rename the lib, as you will see in the commit history once the PR is open. So for the most part your work will be preserved one-to-one.

Also, I'm keeping your Kinetis KE devices as they are. For now, it's really just KL02, KL03, KL04 and KL05 (5 actual symbols in your version), which I designed separate symbols for instead of using as much aliases.

> However, it's open, but we should keep the style of all the NXP MCU symbols consistent, for example with the S08 family.

I think you're right. Symbol styles are indeed quite the same, it's just the pin placement that differs in order to optimize for occupied area in some cases, while keeping a better 'pin compatibility' over different packages within subfamilies where feasible. Following this approach, I've also added KL16 and KL17, KL2x, KL4x as well as some K2x devices so far (see #1540). So it's more of a tweak to the KL0x symbols you designed than it is 'lost' work.

Are you okay with that? :smile:

myfreescalewebpage, 2017-10-03:
@t3hcatpaw I'm comforted now your additions will be great and I agree that it's better to keep Kinetis family consistent. You are going to add many Kinetis if I clearly understand, so with this objective of consistency on the library it's ok for me to replace the few ones I have created of course :-)

Yes I have seen KLC are involved and the library will be renamed. Good to read that contributions are supervised to get Kicad very nice to use - a difficult job when we see all people giving their own piece of the puzzle ! Thanks to the Kicad team :-)

t3hcatpaw, 2017-10-03:
Thank you @myfreescalewebpage! Didn't want to mess your work without your consent :+1: And yes, I'm planning on adding more Kinetis in the near future. Please, feel free to join me :smile:

@SchrodingersGat Opened #1683. :grin:

Best regards,
Dennis

closed 2017-10-05 by t3hcatpaw