KiCad/kicad-library #1684: VSOURCE symbol of pspice.lib is too huge

efferre79, 2017-10-04:
I have just installed kicad and kicad-libraries from the github repositories compiling from source. As a tutorial I am trying to draw a schematic and simulate it with ngspice. I have loaded the pspice.lib (/usr/share/kicad/library/pspice.lib) but the symbol for the VSOURCE is definitely too huge comparing to the other symbols, here is a snapshot:

[image]

Looking at the video tutorial https://www.youtube.com/watch?v=2WDPzW6DGzQ, it looks regular. Is it a problem of the library or kicad?

SchrodingersGat, 2017-10-04:
It does look like it needs to go on a diet. Are you able / willing to submit a pull request to address this?

efferre79, 2017-10-04:
I have just started using kicad so I don't have confidence at all with the software and the libraries, I don't think I am able right now to fix the issue by myself :-( But I am available to test the changes in the library.

SchrodingersGat, 2017-10-05:
I have submitted a change (see above).

SchrodingersGat, 2017-10-05:
This has been fixed now, thanks for bringing it to our attention.

closed 2017-10-05 by SchrodingersGat

efferre79, 2017-10-05:
I have tested the changes, now it looks better but I think it can be made smaller. I am attaching another attachment below where I put the voltmeter symbol (from device.lib) as a reference, in my opinion the circle of VSOURCE can be made as small as the one for the voltmeter.

[image]

What do you think?