KiCad/kicad-library #1722: Kingbright Display library

brekee12, 2017-10-14:
Hi there,

Kingbright make available it's LED display 3D CAD files. I could convert IGES files to DXF but I could not create a part from it. Could you help me how to do it or help to do it.
Thanks in advance!

SchrodingersGat, 2017-10-15:
Not quite sure what you are asking. Do you want to convert the DXF files to a footprint, or use the data to manually create footprints yourself?

brekee12, 2017-10-15:
I would like to use Kinbright ACSC08-51QBWA/D LED display. It is a new SMD
package which has a CAD drawing (IGES) publicly available. I was succesful
to convert it to DXF by FreeCAD and I could import it, but unsuccessful to
make a 3D parts from that. I would like to do it by myself to learn it, if
it is possible; but I have seen 3D packages are on GitHub and they are in
special .pretty fileformat, which are managed by KiCAD team. What would be
the correct methode: request to make this library component by the team or
please suggest material how to do it myself.

bobc, 2017-10-15:
Ok, there is a lot of confusion here. The .pretty file contains the _footprint_ only, but it can also include the file name of a 3D model if there is one.

The 3D model is a completely separate file to the footprint file, and can be VRML 2.0 (.wrl) or STEP format. I suggest reading up on the KiCad documentation will help you understand the process.

The normal workflow is:
1) create footprint (.pretty) file from data in the datasheet
2) create or obtain a 3D model in VRML or STEP format.

brekee12, 2017-10-16:
Thanks for the instructions I succesfully created the parts. The library
structure has to be established, but in the 3D view everything looks OK.

closed 2017-10-16 by brekee12