KiCad/kicad-library #1744: I am creating a DC Power Supply in Kicad. I can't able to clear the error in Kicad EESCHEMA.

Nellairaja, 2017-10-19:
[image]
[image]
If you know how to clear a error. I need to create a footprint for this Dc power supply project. To create a DC POWER SUPPLY in PCB 
Please reply me to this email id : rajatvl27@gmail.com

antoniovazquezblanco, 2017-10-19:
This is probably more suited to be posted in the forums and not library issues but anyways:

From your image you have not properly connected many components. Delete the wires with conflicts and try to wire things from the end of the pin. I've highlighted the conflicting points I could see in your image:
[image]

gauravjuvekar, 2017-10-19:
@Nellairaja  Please post this to https://forum.kicad.info/

closed 2017-10-19 by jkriege2

Nellairaja, 2017-11-02:
> [image: Boxbe] <https://www.boxbe.com/overview> Jan W. Krieger (
> notifications@github.com) is not on your Guest List
> <https://www.boxbe.com/approved-list?tc_serial=33618827386&tc_rand=507836859&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=TcT8ru8I%2FcpPkm2sIjgzHBl6edmYbg4K8QncKz2Bewk%3D&token=jYd0cQfTd5eIB%2FeECRNEAv5Nb1MHanuuDKw6oSY3mYE1OK4%2FwMbn97Vv%2BJdx1nI0>
> | Approve sender
> <https://www.boxbe.com/anno?tc_serial=33618827386&tc_rand=507836859&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&key=TcT8ru8I%2FcpPkm2sIjgzHBl6edmYbg4K8QncKz2Bewk%3D&token=jYd0cQfTd5eIB%2FeECRNEAv5Nb1MHanuuDKw6oSY3mYE1OK4%2FwMbn97Vv%2BJdx1nI0>
> | Approve domain
> <https://www.boxbe.com/anno?tc_serial=33618827386&tc_rand=507836859&utm_source=stf&utm_medium=email&utm_campaign=ANNO_MWTP&utm_content=001&dom&key=TcT8ru8I%2FcpPkm2sIjgzHBl6edmYbg4K8QncKz2Bewk%3D&token=jYd0cQfTd5eIB%2FeECRNEAv5Nb1MHanuuDKw6oSY3mYE1OK4%2FwMbn97Vv%2BJdx1nI0>
>
> Closed #1744 <https://github.com/KiCad/kicad-library/issues/1744>.

In this project I am try to clear the error and reconnect the circuit with
wire... But the wire was not connected properly. I didn't know what is the
reason.... Take me how to connect the circuit with the wire tool

poeschlr, 2017-11-02:
Please post your question over at the user [forum](https://forum.kicad.info/). (You don't even need to create a new account. You can use single login with your github, facebook, google, ... account.)

Nellairaja, 2017-11-04:
please help me to clear the above error in this project.

poeschlr, 2017-11-04:
Again this is the wrong platform for this. Is there a reason why you think the forum is not for you?

The forum has more powerful tools for people who want to help you. 

There is also a more diverse group of people available to help you. 

This issue is mostly only seen by library maintainers. Which means you spam our wall instead of reaching the people who can help you.

sandhan26, 2017-11-05:
@Nellairaja you might have accidentally altered the grid size, while working with the wires. Thats why they are not connecting. 

The best thing for you would be, start fresh, and be mindful of not changing the grid size, don't rush.
Trust me, i used to get a lot that.