KiCad/kicad-library #1747: Move templates to a separate repo?

heitorPB, 2017-10-20:
Hi, all!

So, i am [trying to] create a new template for a new Arduino board and I will create a PR soon. But I saw that the 3D component lib is now in a separate repo from this one. Should the templates move also to a separate repo?

Just to make everything easier to find:

- schmatic lib in a repo
- each footprint lib in a repo
- 3D models in a repo
- templates together with schematics?

SchrodingersGat, 2017-10-20:
The templates will be moved to `kicad-templates` repository as part of the library reorganisation for the v5 release. 

* `kicad-symbols`
* `kicad-footprints`
* `kicad-packages3d`
* `kicad-templates`

SchrodingersGat, 2017-10-20:
Continue to contribute here for now, there will be a transfer of data at some point in the future.

closed 2017-10-20 by SchrodingersGat