KiCad/kicad-library #1762: Two Bosch libs with different names; causes conflict on Windows

bobc, 2017-10-23:
There are two libs for same manufacturer with different names. On Windows, library names are not case-sensitive, so these libs conflict.

Proposal:
- move parts from bosch.lib to Bosch.lib
- remove bosch.lib/.dcm

ppelleti, 2017-10-23:
This is a problem on Mac OS X as well.  It causes git to always think that Bosch.lib/.dcm are modified, so it will be difficult to make any contributions downstream of this commit until it is resolved.

sandhan26, 2017-10-24:
I have a Linux system. I can move the only component 'BMF055' present in 'bosh' library to 'Bosh' library, and delete the 'bosh'. I will take time until evening to fix it.

SchrodingersGat, 2017-10-24:
PR here - https://github.com/KiCad/kicad-library/pull/1767

closed 2017-10-24 by jkriege2

bobc, 2017-10-24:
Yes, I am also stuck with submitting PRs due to this issue!

jkriege2, 2017-10-24:
merged the PR

bobc, 2017-10-24:
Just as I was typing, thanks! 👍