KiCad/kicad-library #1769: problem with CvPcb particularly association of footprint with component through GITHUB/internet

rightwing, 2017-10-24:
I'm a novice in KiCAD so be patient. The issue is:
While following help guide of KiCAD I reached till creating netlist without error.
**When I tried to run CvPcb to associate component and footprints, it tried loading footprints from github. However this couldn't be completed** and after a very long "non-responding" session it told me so. I would like to clarify here that I sit behind a proxy network and I couldn't get to provide username or password option for this step of associating component and footprints. So, (that's what I think) it took a lot of time to give me the error list:

----
_Errors were encountered loading footprints:
IO_ERROR: http GET command failed
Cannot get/download Zip archive: 'https://codeload.github.com/KiCad/TO_SOT_Packages_SMD.pretty/zip/master'
for library path: 'https://github.com/KiCad/TO_SOT_Packages_SMD.pretty'.
Reason: 'IO_ERROR: curl_easy_perform()=7: Couldn't connect to server
from kicad_curl_easy.cpp : Perform() : line 92'
from github_plugin.cpp : remoteGetZip() : line 554_
----

Is there any way I would connect to internet for the same or a simpler way to store it locally.
I'm not sure though if I made myself clear, so please revert back in that case.

rightwing, 2017-10-24:
[Reported problem earlier](https://forum.kicad.info/t/cvpcb-does-not-see-footprints/1246/9)
is what I'm facing. I think.

bobc, 2017-10-24:
KiCad does not have any support for proxy servers, so unless you are able to setup a proxy, you can't use the online libraries in KiCad.
You will need to download the libraries and install them locally, there is a snapshot at http://downloads.kicad-pcb.org/libraries/ (download the latest footprints zip).
I recommend https://forum.kicad.info/ is generally a better place for this type of question.

rightwing, 2017-10-25:
Thanks @bobc for quick and positive reply.

rightwing, 2017-10-25:
I have a doubt though, there aint any .petty file for footprint.
And I'm quite new to KiCAD, can you suggest some link or way to add these library files to local destination and setup the environment for footprint and component association.

poeschlr, 2018-01-20:
@rightwing this is not the right place to ask for support on how to use kicad. I have written a tutorial for footprint lib installation over at the kicad forum https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269. If you still have some questions please ask over at the forum

closed 2018-01-20 by poeschlr