KiCad/kicad-library #1771: Why are the pinheaders 3d-shapes so extremely large?

TotalKrill, 2017-10-25:
Looking at the list in the modules/packages3d/Pin_Headers.3dshapes folder, I see that the folder on my computer is at 1.5GB. That is insane! And looking into it I see that a lot of the stepfiles are actually 3.1MB large in that folder. 

I am guessing that there has been some auto-generation involved here. But maybe this should be fixed?

Shackmeister, 2017-10-25:
The biggest models are around 3 MB yes.. 1.5GB is alot indeed but so is 2200 models.
Im not sure if there is an easy way to make the models smaller, perhaps @easyw has an idea?

bobc, 2017-10-25:
STEP files are not compact, that is a well known feature. A 1 pin connector is about 40kB in STEP (versus 10kB in WRL), so 40 x 80 pins = 3.2MB, and 4 times size of WRL.

Both formats are readable ASCII, STEP is just more wordy. I don't think there is any fix that can be applied to the file size. Workarounds might be use a different file format (but STEP is the standard for mechanical design), or distribute as zip and do some "on demand" dezipping in Kicad, which requires some work at both ends.

In the race between more efficient software and faster hardware, faster hardware usually wins. I have a 2TB disk I bought several years ago, even with multiple copies of kicad repos, I have still only used 30%. I started out writing code on a computer with 1KB RAM, floppy disks stored 360KB, efficient coding was essential. Nowadays my incentive to write efficient code has reduced significantly!

tldr; buy a bigger disk, don't worry about it.

easyw, 2017-10-25:
Hi @TotalKrill 
solution for your size issue:
1. you can avoid to download all 3D models

or

2. you can delete the models you don't need

**The STEP file size isn't 'insane' at all!** You can have a look at some manufacturer and if you check some similar model you will see that the size will be the very similar (in many cases it will be even bigger)
I worked closed to the FC community to improve the STEP importing/exporting and performance
you can search at FC forum
https://forum.freecadweb.org/viewtopic.php?f=9&t=23371
the file size of STEP files has been reduced up to 5 times in many cases and the KiCad library has been developed already with that feature... FreeCAD now is deploying STEP file of really small size compared to other proprietary and not sw.

Maurice

TotalKrill, 2017-10-25:
@easyw Wow, apparently the step size isnt insanely large compared to others, you are right. Its just step that is insanely large in general. And well since it is industry standard I guess it is reasonable to have it in step. I still think having 1.5GB or in my case 1% of my hard drive and 33% of my data plan for step files of pin headers is a bit overkill.

I also noticed that there is a packages3d library as well. Guessing this means that the 3d models are going to be removed from here entirely?

 I just find it a bit strange that half of the symbol library total size is step models of basically 3-4 different pin types in varying amounts. And that is without the  the 3 row versions of pin headers common in RC. 

Why isnt having the different pin models just added multiple times to the footprint an option? This way we would have 3-4 different step models just iterated a few thousand times. 

If I would apply a script to fix this, would that get rejected? 

@bobc buying an ssd of that size to fit my laptop isnt really feasible. And I am mostly annoyed since I now have exactly no data left of my mobile data ( yes i know its my mistake ).  Luckily only for a few days.

SchrodingersGat, 2017-10-25:
> I also noticed that there is a packages3d library as well. Guessing this means that the 3d models are going to be removed from here entirely?

Yes the symbols and 3Dmodels are going to be split

> Why isnt having the different pin models just added multiple times to the footprint an option? This way we would have 3-4 different step models just iterated a few thousand times.

Arraying 3D models would be cool! Perhaps add the following parameters to a model association:

* array_x
* step_x
* array_y
* step_y

So e.g a 1x10 x 2.54mm pin header:

* array_x = 10
* step_x = 2.54

SchrodingersGat, 2017-10-25:
BTW downloading the `.tar.gz` compressed archive of the `kicad-library` repo is only ~550MB

TotalKrill, 2017-10-25:
@SchrodingersGat Yeah,  It would be cool! But I would be fine with just scripting the addition into the footprint files, since that would mean no changing of file formats etc etc. 

And glad to hear they will be split! 👍

bobc, 2017-10-26:
> Why isnt having the different pin models just added multiple times to the footprint an option? This way we would have 3-4 different step models just iterated a few thousand times.

Interesting idea, I've used that method with WRL to create pin headers when I didn't have the right model available.

The primary purpose of STEP is for export to mechanical design package, not sure how having multiple pins instead of a single package will play there.

Shackmeister, 2017-10-26:
I suppose it could work, but would be kinda annoying for MCAD stuff. I can probably make the change but would like to hear other opinions, especially from @easyw

SchrodingersGat, 2017-10-26:
I would like to see if KiCad can merge the STEP files together and then export a single MCAD blob. 

Then you could do some pretty cool things. Say you wanted a pin-header where only every second pin was fitted, just add two models:

1. Plastic piece
* repeat-x : 10
* spacing-x: 2.54

1. Metal pin
* repeat-x: 5
* spacing-x: 5.08

Shackmeister, 2017-10-26:
It does use OpenCascade...for stepup I suppose it would be quite easy to make it fuse all assigned models within a footprint. Same could probably be done for kicad. If arrays were supported

SchrodingersGat, 2017-10-26:
I might spend some time investigating whether such a feature would be possible

easyw, 2017-10-26:
what you need is a STEP hierarchy with external step models... I mean you build a 36pins header linking 1pin 36 times... so your STEP file contains only 1pin and the info for placement of the other 35pins...
but:
1) this is NOT supported by FreeCAD ATM for STEP files and I don't think it will be supported soon...
2) this is NOT the standard adopted by manufacturers for electronic parts nor by any other ECAD I know around ...
I wouldn't suggest to go in this direction

TotalKrill, 2017-10-27:
Do we know of any other software package that can to hierarchical step models the way you describe?

Is stepz an alternative? I also noticed there are no scripts for the pin-headers in the source version. 

I also dont think git is the correct medium for finished 3d-models, a comparison would be to compile kicad beforehand, and then using git for version control of the binaries, I understand having the source files in git so they can be built locally. But anyway thats oftopic here at least.

TotalKrill, 2017-10-31:
@SchrodingersGat Did you investigate?

Even if the arrays is not supported, one could just use the component path to determine if it is the same component and then merge into a blob for exporting. 

I would like this option at least.

Misca1234, 2017-11-05:
For common knowledge

If you use the "grid" utility in FreeCad, that is, have one single component and the "array it", 
which is somewhat a natrual utility to use when you create one leg and then "array it" to get an easy way of setting distances 

[image]

The step file size increase in size dramatically, much more than expected

SchrodingersGat, 2017-11-05:
I now have the "array" feature working well in KiCad. You can specify a "repeat" parameter and the offset in x/y/z axes.

* Displays properly in 3D renderer
* Displays properly in rayrtracing renderer
* Outputs VRML with multiple references to the same model
* Outputs STEP correctly

Surprisingly, the generated STEP file can be much much smaller.

e.g. export a PCB with a single footprint of a 2x33 pin header, one with the original model and one with PinHeader_1x01 model arrayed 33 times:

* Size with single model: 6.3MB
* Size with 33x individual models: 349kB

SchrodingersGat, 2017-12-11:
@TotalKrill I'm going to close this issue as we are now splitting the symbols and footprints into separate repositories.

closed 2017-12-11 by SchrodingersGat