KiCad/kicad-library #1772: API for eeschema

ZhukovGreen, 2017-10-25:
Please advice,

We have to be able programmatically:
1. Modify text on schemes
2. Join schemes into one scheme
3. Print compound scheme

Then we have to have a database of components which we may drug and drop into the scheme. Then:

1. Calculate the overal footprint
1. BOM
1. Cost

Is it possible?

bobc, 2017-10-25:
There seems to be a trend for using the Issues tracker as a general support forum, let's keep it for reporting issues with the library components and footprints please?

The recommended forum for general KiCad issues is https://forum.kicad.info/

SchrodingersGat, 2017-10-25:
@ZhukovGreen please direct this question to the [user forums](https://forum.kicad.info) as Bob suggested :)

closed 2017-10-25 by SchrodingersGat

ZhukovGreen, 2017-10-25:
OK, sorry guys.
/close:)