KiCad/kicad-library #1776: The following libraries were not found in kicad

www2000, 2017-10-28:
Hi i wand to report that in the ubuntu build of kicald librart that is trlease on 2017-10-27 at 08:49(i beleave gmt) missing some libarys:
`
The following libraries were not found:
transistors.lib
conn.lib
74xx.lib
cmos4000.lib
display.lib
valves.lib`

poeschlr, 2017-10-28:
Is this a nightly build? If so than this is to be expected. We have already started to reorganize the lib in preparation for kicad v5. (And the nightly builds ship the libs of the same night. Your local symbol lib lists are now out of sync)

These libs have been renamed. the conn lib is now called Connector, the 74xx lib and cmos lib now have the prefix Logic_ and the transistor lib is now called Transistor. 
Some other libs have also changed their name since then.
More details about the reorganization see: https://github.com/KiCad/kicad-library/issues/1402

www2000, 2017-10-28:
Thanks i have no problem with the that the symbol lib point to transistor.lib and in the my filesystem use Transistor.lib the the capital T in state of a lower case t

aurabindo, 2017-11-10:
In the git version of the libraries, some lib files mentioned in `templates/sym-lib-table` have not been updated to corresponding files in `Library/`, like stale references to `cmos_ieee.lib` instead of `Logic_CMOS_IEEE.lib`

evanshultz, 2017-11-10:
I just merged https://github.com/KiCad/kicad-library/pull/1818, which will hopefully get things aligned for now. We apologize for any potential and periodic breakage until v5 is released due to library reorganization. Thanks for reporting this issue and using KiCad!

SchrodingersGat, 2017-12-11:
@www2000 the new [kicad symbols library](https://github.com/kicad/kicad-symbols) will track the sym-lib-table file to make sure that it does not get out of step again

closed 2017-12-11 by SchrodingersGat