KiCad/kicad-library #1787: Incorrect pin electrical type on 74LS30.

117L7, 2017-11-01:
The 74LS30 from the 74xx library has the output pin (8) marked as an input pin. This causes ERC failures.

KiCad 4.0.7 for OS X.

SchrodingersGat, 2017-11-02:
@117L7 can you submit a PR to address this issue? This will be the quickest path to fixing :)

117L7, 2017-11-02:
Sure, how would I do that?

SchrodingersGat, 2017-11-02:
http://kicad-pcb.org/libraries/contribute/

If this all seems too much for you, perhaps someone will have time to make the fix for you.. But, if you are willing to learn, extra contributors are *always welcomed!*

bobc, 2017-11-02:
I think I already fixed this. Did you check the latest version of Logic_74xx ? It looks correct to me.

117L7, 2017-11-02:
I downloaded kicad-library-master, it looks like the libraries have been
reorganized compared to the 4.0.7 distribution.

I'll give Logic_74xx a try later today.

poeschlr, 2017-11-02:
Yes we have already started the preperation for the v5 release. You might want to use some older commit.
@jkriege2, @SchrodingersGat should we tag the last commit before we started with the reorganization?

117L7, 2017-11-02:
I tried Logic_74xx, but the problem now is the symbols are a smaller size (which is a good thing), and the shrinkage apparently caused all the devices to disconnect from the nets. I didn't expect this would happen.

I uploaded some screen shots before/after, from the original 2 to now 299 ERC errors, so the testing is inconclusive.

I was able to restore by removing the library and using the schematic redraw button.

[image]
[image]

poeschlr, 2018-01-09:
Not only the size did change. We also moved the power pins to a separate unit. (They where invisible pins placed at every unit in the old lib.)

This is why we will release this stuff only for the v5 release. Users will expect massive changes on a major release so the surprise will not be that bad.

poeschlr, 2018-01-20:
If the electrical types are still wrong please reopen this over at https://github.com/KiCad/kicad-symbols. Regarding recising: you will need to accept these changes as the new lib is intended for a new major release and we allow major changes there. We will try to keep the lib as stable as possible during the next stable release cycle.

closed 2018-01-20 by poeschlr