KiCad/kicad-library #1788: Schematic symbols convention

El3ctr0nic, 2017-11-02:
Hi All. I don't know if it is good place for this discussion.

KiCad symbols in libs are changed frequently. I think that some of them should not be changed (in my personal opinion). Example below TLP250 was similar to ACPL (btw My drawing). Now it's broken/reworked i don't know. Some convetion changed??

[image]

I want help libs/symbols migration but im a little bit confused. I don't know symbols visual style direction.
Only pin names symbols without graphic inside are small as possible but schematic is very monotonic.

We have drivers_mosfet.lib schoulden't we split it to isolated and nonisolated ??
ACPL and TLP schould be moved to drivers_mosfet i want do this but how to rework them??

Pls feedback :) 

Regards,
Marcin

poeschlr, 2017-11-02:
About lib organization have a look here: (will happen for the v5 release)
https://github.com/KiCad/kicad-library/issues/1402

About graphical guidelines: Well we try to keep it consistent within a lib. (This means if there are a few symbols that look different they should be changed such that they all look similar.)

poeschlr, 2018-01-20:
@El3ctr0nic we will try to keep libs more stable in future. Feel free to reopen this in the new repo https://github.com/KiCad/kicad-symbols if we do something stupid again

closed 2018-01-20 by poeschlr