KiCad/kicad-library #1799: Slash in symbol names - violation of the new KLC?

lavadew, 2017-11-05:
Hi there,

i have a question regarding the naming of symbols.

For example, there is a PIC microcontroller named "PIC10(L)F320-I/MC".

But when i check the lib with the kicad-library-utils, it says that the slash between I an MC is a violation of the KLC.

Should i change the name of the symbol?

Best regards,

lavadew.

SchrodingersGat, 2017-11-05:
The `(` and `)` characters are also violations!

These symbols predate the new requirements, which is why they are still in the library.

In short, yes if you are submitting new symbols they must not have these characters.

e.g.

* `PIC10LF320-IMC`
* `PIC10F320-IMC`

bobc, 2017-11-05:
I did a grep and there are quite a few :)

```
adc-dac.lib:DEF MCP3425A0T-E/CH U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP3550-50-E/MS U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP3550-60-E/SN U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP4801-E/MC U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP4901-E/MC U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP4921-E/MS U 0 40 Y Y 1 F N
adc-dac.lib:DEF MCP4922-E/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10(L)F320-I/MC U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10(L)F320-I/OT U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10(L)F320-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F200-I/MC U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F200-I/OT U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F200-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F204-I/MC U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F204-I/OT U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F204-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F220-I/MC U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F220-I/OT U 0 40 Y Y 1 F N
MCU_Microchip_PIC10.lib:DEF PIC10F220-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12(L)F1501-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12(L)F1822-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12(L)F1840-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C508-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C508A-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C509-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C509A-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C671-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12C672-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12CE518-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12CE519-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12CE673-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12CE674-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12CR509A-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F508-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F509-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F510-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F519-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F609-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F615-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F617-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F629-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F635-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F675-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F683-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12F752-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12HV609-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12HV615-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12HV752-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC12.lib:DEF PIC12LF1840T48-I/ST U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1454-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1454-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1455-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1455-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1459-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1459-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1503-I/MG U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1503-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1507-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1507-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1508-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1508-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1509-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1509-I/P U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1512-I/MV U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1512-I/SP U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1513-I/MV U 0 40 Y Y 1 F N
MCU_Microchip_PIC16.lib:DEF PIC16(L)F1513-I/SP U 0 40 Y Y 1 F N
[... 107 lines cut ...]
MCU_Microchip_PIC18.lib:DEF PIC18F4455-I/PT U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F4458-I/ML U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F4458-I/P U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F4458-I/PT U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F448-I/P U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F44J10-I/P U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F4580-I/P U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F45K80-I/ML U 0 40 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F45K80-I/PT U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F66J60-I/PT U 0 50 Y Y 1 F N
MCU_Microchip_PIC18.lib:DEF PIC18F96J60-I/PT U 0 20 Y Y 1 F N
MCU_Microchip_PIC24.lib:DEF PIC24FV32KA304-I/PT IC 0 40 Y Y 1 F N
MCU_Microchip_PIC32.lib:DEF PIC32MX1XXFXXXD-I/PT U 0 40 Y Y 1 F N
MCU_Microchip_PIC32.lib:DEF PIC32MX2XXFXXXD-I/PT U 0 40 Y Y 1 F N
microchip.lib:DEF MCP2050-330-E/MQ U 0 40 Y Y 1 F N
microchip.lib:DEF MCP2050-330-E/P U 0 40 Y Y 1 F N
microchip.lib:DEF MCP2050-330-E/SL U 0 40 Y Y 1 F N
microchip.lib:DEF MCP2515-E/ST U 0 40 Y Y 1 F N
motorola.lib:DEF MCF5328/BGA256 U 0 40 Y Y 1 F N
sensors.lib:DEF MCP9700AT-E/LT U 0 40 Y Y 1 F N
sensors.lib:DEF MCP9700T-H/TT U 0 40 Y Y 1 F N
xilinx.lib:DEF XC3S1400A/FG484 U 0 40 Y Y 2 L N
xilinx.lib:DEF XC3S200AN/FT256 U 0 40 Y Y 1 F N
xilinx.lib:DEF XC3S50AN/TQG144 U 0 40 Y Y 1 F N
```

lavadew, 2017-11-05:
okay, we have quite a bunch to change.
thanks for the info, issue can be closed.

SchrodingersGat, 2017-11-05:
If you have the time, a fix for this would be greatly appreciated :)

lavadew, 2017-11-05:
yeah, will tackle this.

poeschlr, 2017-11-05:
I would say we don't close this until the pull request is created.

bobc, 2017-11-05:
I think we can leave open, it is a useful placeholder.  I used the one line 
`egrep "^DEF" *.lib | egrep -v "DEF [~]*[A-Za-z0-9-_\.]+ " >t.txt

This found some parts using "+", including many power symbols of course e.g. "+5V" -  should "+" also be allowed in name?

In addition to parts using "/":
```
device.lib:DEF D_Bridge_+-AA D 0 50 Y Y 1 F N
device.lib:DEF D_Bridge_+A-A D 0 50 Y Y 1 F N
device.lib:DEF D_Bridge_+AA- D 0 50 Y Y 1 F N
device.lib:DEF D_Bridge_-A+A D 0 50 Y Y 1 F N
device.lib:DEF D_Bridge_-AA+ D 0 50 Y Y 1 F N
maxim.lib:DEF DS1307+ U 0 30 Y Y 1 F N
maxim.lib:DEF MAX7325AEG+ U 0 40 Y Y 1 F N
motor_drivers.lib:DEF L298(H)N U 0 40 Y Y 1 F N

SchrodingersGat, 2017-11-05:
@bobc good point regarding the `+` character, had not thought of this one. We definitely need to allow that.

lavadew, 2017-11-08:
- [ x] PR for adc-dac
- [ ] PR for MCU_Microchip_PIC10
- [ ] PR for MCU_Microchip_PIC12
- [ ] PR for MCU_Microchip_PIC16
- [ ] PR for MCU_Microchip_PIC18
- [ ] PR for MCU_Microchip_PIC24
- [ ] PR for MCU_Microchip_PIC32
- [ ] PR for microchip
- [ ] PR for motorola
- [ ] PR for sensors
- [ ] PR for xilinx

lavadew, 2017-11-08:
question for the Microchip MCUs: Everyone agrees, that the LF variants should be aliases?

SchrodingersGat, 2017-11-08:
Yes using aliases is a sensible approach.

hildogjr, 2017-11-12:
Yes, I had the problem with the MCP4922-E/P too.
This problem just appear in the 10/Nov/2017 update (maybe because the new table library in Eeschema).
To turn around I create my on symbol "MCP4922-E_P" copying the internal part in the library to may personal library.

tr-soft, 2017-11-29:
I'm sorry, but can I found somewehere why you choose to disallow slash in part names? I'm software developer too, but I can't imagine that I make a decision where I disallow naming parts in form in which manufacturer labels them. It doesn't make any sense that developers of software to aid create schematic "dictate" me that I've to name parts differentially.

poeschlr, 2017-11-29:
Well first off all the symbol/footprint names are used as file names. (for symbols this will happen shortly as soon as the new file format is published.) "/" in filenames is a bad idea because this char is use as path separator.

And it seems that the current implementation of the new symbol lib table already assumes there is no such char in the names.

tr-soft, 2017-11-29:
Ok, I see the parallel with filenames but this is developer's decision that component have to have same name as filename and component name is "duplicated" in file content. Other tools doesn't have this constraint, I think this is as same as if OS developers says that 'red' color will not be supported in new version. In the market there are many many parts that have slash in label, so why make decision that this wouldn't be possible.

poeschlr, 2017-11-29:
Ask over at the mailing list ;)

bobc, 2017-11-30:
The decision to store data in separate files was made a while ago, personally I would not have done it that way, but I don't have influence on the developers. Even so, the file name could be suitably encoded to ensure that a valid file name is created - cf encoding of HTTP urls. The file name might lose a little readability if special chars are used, but not much. 

Regardless, any scheme different to the current one requires someone to write some code, and that doesn't happen by magic. You could either discuss with the developers and submit a patch, or raise an issue on the bug tracker and hope that one day a developer shares your point of view and is motivated to improve KiCad.

SchrodingersGat, 2017-12-11:
I am closing this issue as all the illegal characters have been fixed in the [new symbol library](https://github.com/kicad/kicad-symbols)

closed 2017-12-11 by SchrodingersGat