KiCad/kicad-library #1814: Triac symbols needs correction, or new symbol

Samonitari, 2017-11-08:
Triac symbols does not differentiate between A1 and A2 (or M1 and M2 as **better known**) terminals.
While historically that was okay, nowdays it is recommended that you use a "III quadrant triac" if possible, so snubber network can be left out. These triac does not behave symmetrically.

Problem is: take for example "Q_TRIAC_AAG" symbol. Conventional quadrant diagrams draw that when gate 'curls' down on left side (normal orientation of symbol) upside terminal is M2. Symbol uses that as PIN1.
Use that with for example this :[T435 triac in TO-252](http://www.st.com/content/ccc/resource/technical/document/datasheet/cf/95/c3/9d/a2/b7/47/43/CD00002383.pdf/files/CD00002383.pdf/jcr:content/translations/en.CD00002383.pdf)
PIN 1 and 2 are **NOT** fully interchangeable, and PIN1 is A1 phisically, but A2 on symbol.
It could cause PCB-s going to the trash!

jkriege2, 2017-11-19:
Hi @Samonitari !

do you mean symbols like these?
[image]
[image]

I couldn't really find a reference on the internet and am not so familiar with TRIACs ... maybe you can also simply post a PR that renames/adds the symbols appropriately?

Best,
JAN

poeschlr, 2017-12-04:
@Samonitari could you look at the suggestion by @jkriege2? I would really like to see this resolved for kicad v5 (So until end of December)

Samonitari, 2017-12-04:
I am sorry for not replying!
(I was behind schedule in PCB designing)
Briefly: that's what I wanted, but I gladly make a pull request this week!

Samonitari, 2017-12-04:
Hi @jkriege2 !
After really inspecting the proposition, is too much. I think I was not clear enough on the problem, excuse me.

I did not see a manufacturer datasheet that draws its triac symbol the way the gate is nearer to M1, but I'll have a look around more deeply. But i thinks it is just a (maybe unwritten) convention: gate curls down, and M2 side is up (in normal orientation). I do not want to mix things up unneccesarily. or to create distractions.

So my pull request will only contain Q_TRIAC_M1M2G and like symbols. the only added value being the possibility to differentiate between M1 and M2, effectively doubling the basic triac symbols.

Is it okay with you, and everybody?

evanshultz, 2017-12-05:
I believe the first part of this issue is that the pinout of the symbols doesn't map to all the footprint. Looking at the datasheet linked above, we see this:
- Pin1 = A1
- Pin2 = A2
- Pin3 = G

That would mean a symbol `Q_TRIAC_A1A2G`, to be clear about A1 and A2. We have `Q_TRIAC`AAG` which is already correct, but the symbol name is unclear. Resolving this issue will require more symbols since we don't have all permutations of pin ordering and we can't be sure of the pin order in the footprints.

The second part is that our symbols put A2 near G instead of putting A2 opposite the gate. This is just the symbol convention and seems fine to me to change all symbols.

@Samonitari Is that what you had in mind? I'm fine with the above two changes. All existing symbols will be replaced, so I think this happens only in the v5 library. So we will have 6 symbols, all with A2 (or M2) facing up and G facing down, and all pin arrangements. It's probably easiest if the pin names are shown along with numbers then. Anybody else object to this?

And while I was looking around in here, I noticed there are symbols called `Q_THYRISTOR*` which are actually SCRs, a type of thyristor. TRIACs and DIACs and other parts are also thyristors. I believe those symbols should be renamed to `Q_SCR*`.

poeschlr, 2018-01-20:
I will close this here as we transferred over to a new repo. I reopened this issue over there: https://github.com/KiCad/kicad-symbols/issues/182

closed 2018-01-20 by poeschlr