KiCad/kicad-library #1826: IEC 74xx symbols

mrnuke, 2017-11-14:
I'm working on IEC symbols for the 7400 series [here](https://github.com/mrnuke/kicad_iec_symbols).

I'm trying to figure out how to integrate this the existing library. In the master branch, I only have those symbols which have been production tested. i.e. the schematic was laid out, a board was manufactured, assembled and verified to be working.

What's the best way to go about incorporating this?

SchrodingersGat, 2017-11-14:
See notes in your PR :)

bobc, 2017-11-14:
I did quite a lot of work on my scripts to generate IEC style symbols, but had not quite finished. As the scripts became larger I had to do some essential refactoring.

I started with cmos_ieee because there were fewer symbols in that lib. A typical example of script generated symbol:

[image]

The script generates control block, elements, can also handle grouped inputs. There is some automatic pin decoration of tristate pins, also ability to label a pin, a group or an element. It turns out most of the required special symbols are in the Unicode font already.

Here is a draft of a complete set.

[demo_cmos_ieee.pdf](https://github.com/KiCad/kicad-library/files/1470166/demo_cmos_ieee.pdf)

SchrodingersGat, 2017-12-11:
@mrnuke please readdress this to the new kicad-symbols repository :)

closed 2017-12-11 by SchrodingersGat