KiCad/kicad-library #1827: Separate 3d modules from schematic symbol libraries

mrnuke, 2017-11-14:
```
du -sch *
4.0K    CMakeLists.txt
8.0K    CMakeModules
4.0K    CONTRIBUTING.md
28K     COPYRIGHT.txt
16M     library
4.0K    library-check.sh
3.1G    modules
4.0K    README.md
6.3M    template
3.2G    total
```
The 3D modules are huge, and bog down the repo. It doesn't make sense to me that these two would be put in the same repo.
Why does someone need to pull in the 3D modules if they want to work on a symbol? It doesn't make sense. It's madness.

SchrodingersGat, 2017-11-14:
Madness is exactly what it is.

We are currently working on transferring all the kicad library data to new repositories:

* https://github.com/kicad/kicad-symbols
* https://github.com/kicad/kicad-footprints
* https://github.com/kicad/kicad-packages3d

There is also now a website where you can download zipped 3D models: https://kicad.github.io/pacakges3d

SchrodingersGat, 2017-12-11:
@mrnuke I am closing this issue, as we have now started the process as described above. Please refer to the new repo URLs

closed 2017-12-11 by SchrodingersGat