KiCad/kicad-library #1833: fp-lib-tables out of date

diggit, 2017-11-15:
Hi, I just noticed, that fp-lib-table* in template directory does not contain all FP libraries and some of them are no longer valid. (89 vs 114 entries)
Is there some automated way how to update them or is it done manually?

diggit, 2017-11-15:
I was not following all recent changes and there is new (for me)  kicad-footprints repo. Will all individual .pretty repos be moved there?

SchrodingersGat, 2017-11-15:
Most of the `.pretty` repos will be moved across. The only ones that are not going to be moved are the ones marked as `deprecated` (i.e. they have been previously been merged with other libs).

Here is the current state of what has been moved across: https://github.com/KiCad/kicad-footprints/blob/master/TODO.md

I need to write some scripting to ensure that the `fp-lib-table` and `sym-lib-table` files are kept up to date.

Can you show me which entries are currently incorrect?

diggit, 2017-11-16:
Not existing repo records in fp-lib-tables are for:
- RF_Antennas
- Connectors_Terminal_Blocks

Until all .pretty repos are merged to kicad-footprints, fp-lib-table should point to individual ones. Am I right?

I was thinking about some script and the only problem is how to get repository name from github...

poeschlr, 2017-11-16:
The github plugin will no longer work once we have one big repo with all fp libs.
This means in v5 the libs will be part of the installation or a separate package.

SchrodingersGat, 2017-11-18:
The new library will keep the fp-lib-table which will keep track of the libraries

SchrodingersGat, 2017-12-11:
Closing this, the new repository is well under way and the sym-lib-table file is checked as part of each PR

closed 2017-12-11 by SchrodingersGat