KiCad/kicad-library #1839: KLC-modifications/additions

jkriege2, 2017-11-17:
Hi @SchrodingersGat @evanshultz @pointhi @poeschlr !

I would like to suggest two additions/modifications to KLC, in part inspired by the discussion https://github.com/KiCad/kicad-library/pull/1828:
1. On S3.8 (http://kicad-pcb.org/libraries/klc/S3.8/). For logic and OPAMPs that's a bit inconsistent. I would suggest to ALSO separate the Power-ports for two special cases, namely logic and OPAMPs (I think these are also the most common devices that come as multi-unit-symbols). My rationale would be: If I design a big logic-circuit, it's good to collect all the power-units at one location and connect them to the appropriate power-plabes (+caps and what not). This tidies up the schematic significantly. If we discourage extra power-symbols for single-unit logic-symbols, this will lead to a mixture, as some logic ICs have their power-pins inside the schematic (may be annoying if you change  power-suuply to them, as you have to do it for every device separately) and some are collected somewhere. Also this would allow us to make unify the single-gate-logic families. Maybe (maybe!) it is possible to design the power-symbols in a way that they can be positioned over the actual symbol (at least for gates), so they can (if wanted) be "combined" to a single device. I think @bobc has some examples of that as alternative symbol style.
2. In addition I would like to see the same exception for OPAMPs for the same reason and with the same style requirement of beeing combineable (I think that ios already common for many multi-unit OPAMPs.
[image]
3. I think we should encourage people to set pin-names for all pins, with only a few exceptions:
  - simple symbols like resistors, caps, ... IF non-polarized (i.e. not for diodes, elcos, ...)
  - logic-gates and output-pins of OPAMPs (there is simply no space)
  In a KLC-check these should light up as warnings, bt not as errors, as the exceptions above cannot really be caught by scripts.

What do you all think?
Best,
JAN

evanshultz, 2017-11-17:
For #1 and #2, I totally agree. I argued before for this and I can't remember why it wasn't accepted. I think separate units for power are all or nothing. And KiCad should be in the all camp.

For #3, I think it's a case-by-case basis. "Encourage" with warnings is a good compromise to me. We will want to incorporate the pin naming styles discussed at https://github.com/KiCad/kicad-library/pull/1611 to allow names and numbers on opamp pins. I like the first option but I can see both ways might be best for different parts. It's a bit annoying the offset can't be selected on a per-pin basis.

jkriege2, 2017-11-18:
@SchrodingersGat What do you think?

evanshultz, 2017-12-06:
@SchrodingersGat @poeschlr Ping!

SchrodingersGat, 2017-12-06:
I'm open to it - can you suggest a new wording for http://kicad-pcb.org/libraries/klc/S3.8/ ?

poeschlr, 2017-12-06:
I like the general idea of the suggestion in this issue. I am not sure how the current KLC would disagree with such symbols. (I think they are already accepted.)

I would not force this on all symbols. Single unit symbols should stay single unit symbols. (The handling of multi unit symbols in kicad is not that good. There is for example no ERC error if you don't place all units of a symbol.)

evanshultz, 2017-12-06:
@jkriege2 I don't see S3.8 discriminating between logic and opamps. It would seem to apply to them both. What is the wording that you dislike?

Perhaps this is a good time to discuss the power symbol design. Does it have a box? Just floating pins (my preference and it also works best if overlaying the gate and power pins as done above). If a box, how big is the box? Is the box filled? Where are pin numbers and names placed around the power pins?

Hmm... no ERC to catch missing units is a problem. Does anybody know if this is in v5 or coming later? Regardless, our multi-unit symbols are already suffering from this issue so why would that stop the single-gate symbols? It does increase the number of symbols that are broken in this way, but I support updating KLC for this and then doing the work when time allows after v5.

jkriege2, 2017-12-07:
This is already a few weeks old ... so maybe I don't remember every detail from then. I think it cooks down to this:

- For OPAMPP and Logic: ALWAYS separate power-pins, so these are consistent over the whole family even if there is only a single device in the package (single OPAMPs / single GATEs)
- I think then having an image showing logic+OPAMP would be great!

About the power-symbol design: I would prefer simply two lines (quite common in other ECAD-programs) + they can easily be combined with the actual symbol. Maybe we should play around a bit with the size to match all possible gates and OPAMPs ... do we need maybe different sizes for the two?

JAN

poeschlr, 2018-01-20:
@jkriege2 if this is still relevant please reopen it over at the new repo. Add a short summary and link back to this issue

closed 2018-01-20 by poeschlr