KiCad/kicad-library #1843: Wrong transistor symbols

bbrandst, 2017-11-18:
The schematic symbols of some transistors in Transistor.lib are not correct.
For instance, both BF244 and BF245 are actually N-JFETs, but in current Git, BF244 is shown as a bipolar transistor while BF245 is presented as a MOSFET. Possibly other transistors are affected as well.
I'm not quite sure but I think this happened during commit 0b9d383.

poeschlr, 2017-11-18:
Mistakes are always possible. (We librarians are not all knowing.)

Car to fix it?

jkriege2, 2017-11-18:
Hi!

I'm already at it (+ some more fixes in that lib) ... might take 1-2 days, but a fix will be posted ;-)

cu
JAN

jkriege2, 2017-11-18:
fixed in here: https://github.com/KiCad/kicad-library/pull/1844

SchrodingersGat, 2017-11-18:
Fixed in #1844

closed 2017-11-18 by SchrodingersGat