KiCad/kicad-library #1851: Library names

DerekGn, 2017-11-22:
Hi,

I have a few library parts that I would like to submit to kicad.

**HDC1080**
A Texas Instruments temperature humidity sensor

**RN4871**
A microchip BLE module

**AAT1217ICA** 
A skyworks  Micropower Synchronous Step-Up Converter

I am looking for some hints with respect to naming convention for the libraries before I submit the pull requests.

Following the conventions I have come up with 

Sensor_Temperature_Texas_HDC
Wireless_Ble_Microchip_RN
Regulator_Boost_Skyworks_AAT

Derek

poeschlr, 2017-11-22:
https://github.com/KiCad/kicad-library/issues/1402

closed 2017-11-22 by DerekGn

SchrodingersGat, 2017-11-22:
@DerekGn please note that you should not place symbols in their own libraries.

I would suggest using more generic names such as 

* Sensor_Temperature
* RF_Bluetooth