KiCad/kicad-library #1863: Some pin headers 3D representation is rotated 90º

fmgomes, 2017-11-30:
I've noticed that with two footprints I'm using, the Pin_Header_Straight_1x16_Pitch2.54mm and the Pin_Header_Straight_2x03_Pitch2.54mm but there are others with the same issue, in 3D view the component is rotated 90º when compared with the footprint. Is there any workaround for correcting that, having a good 3D representation? 
[image]

Best regards

Fernando

Shackmeister, 2017-11-30:
Hello please make sure your footprints and 3D models are updated (don't use packages3D or kicad-footprints, they under renovation at the moment)

fmgomes, 2017-12-01:
Thanks for the help, and what should I use as a source for the update? Where can I find footprint and 3D compatible representations?

Best regards

Fernando

poeschlr, 2017-12-01:
The best way to get compatible footprints/3d models is to download the same release.
All 3d models for the 4.x release series are in the kicad-libary repo: https://github.com/KiCad/kicad-library/releases
Footprints are more complicated. (Each footprint lib has it's own repo)

For your pin headers: https://github.com/KiCad/Pin_Headers.pretty/releases

For further help please refer to this forum post: https://forum.kicad.info/t/incorrect-component-models-in-3d-viewer/7011/3

closed 2017-12-03 by Shackmeister