KiCad/kicad-library #1864: INA138/INA168 Schematic symbol has incorrect pinout (again)

danwood76, 2017-12-01:
Hi,

This is a repeat of the previous issue I raised with the INA138/INA168 pinout (Issue: #1631). I recently re-opened the project that I used the INA138 in to find it wanted to change the pin out back to the original broken version.

I double checked and the fix that I proposed (and that was accepted) appears to have been reverted by commit c64c1d6.

The pinout of the device is clearly documented and hasn't changed, why was my change reverted?

Regards,
Danny

bobc, 2017-12-02:
I think that was just a screw up with someone's branch. Sometimes the history gets confused, and an accidental change gets confused with a good commit. People then incorrectly revert the good commit...

I guess the thing to is to recreate a pull request.

poeschlr, 2018-01-20:
@danger89 a better option than opening an issue is to fix it and open a pull request.

As we already transitioned over to the new repo i will close this here. I opened a new issue over at the new repo: https://github.com/KiCad/kicad-symbols/issues/181

closed 2018-01-20 by poeschlr