KiCad/kicad-library #1870: Add generic IC types.

wgaylord, 2017-12-09:
So you can still you KiCAD with out having to learn how to make a new IC for the library if your IC is not i nthe Library.

(Example LM75BD Temperature sensor which is a SO8)

bobc, 2017-12-14:
Do you mean something like this?

[image]

There used to be some DILxx symbols, but I think they were dropped.

bobc, 2017-12-14:
Just for fun, I created a library with common IC sizes. 

[IC_Generic.zip](https://github.com/KiCad/kicad-library/files/1559133/IC_Generic.zip)

Btw, issues regarding symbols should probably now go to the new repo https://github.com/KiCad/kicad-symbols

poeschlr, 2018-01-20:
I close this as we already transferred all symbols over to the new repo. If this is still relevant please reopen it over at https://github.com/KiCad/kicad-symbols

closed 2018-01-20 by poeschlr