KiCad/kicad-library #1882: Wrong footprint for TXB0108DQSR in library "texas"

troisiemetype, 2017-12-29:
As the title says, the footprint for this component is wrong. The left side (pins from 1 to 10) is for a DQS (USON) package, and the left side for a PWR (TSSOP) package.
The datasheet from TI, see page 3: http://www.ti.com/lit/ds/symlink/txb0108.pdf

I wish I had seen it before I order three prototypes of a PCBA using this part... :(

poeschlr, 2017-12-29:
The footprint is correct (for the component with that order number)

However you are right the pin numbers are wrong. This should be fixed in the new repo https://github.com/KiCad/kicad-symbols

(There will not be a further release using this legacy repo)

So please open a new issue over at the correct repo. And please make sure the title says something about wrong pin numbers not wrong footprint. (Again: the footprint field is correct for this part number)

troisiemetype, 2017-12-30:
Thank you for your answer.
I'm sorry for the wrong title. You're right, the footprint is ok, the pin numbers are not. Shall I Edit the title?
I don't see this chip in the repo you gave the link for; Is it intended to include it at some point? A new issue for a component not to be included would be superfluous...

poeschlr, 2017-12-30:
The texas lib should have been transferred already. (This was the pull request responsible: https://github.com/KiCad/kicad-symbols/pull/109)

poeschlr, 2018-01-20:
@troisiemetype if the problem persists in the new lib, please open a new issue

closed 2018-01-20 by poeschlr